opendr-eu/opendr #103: Non-deterministic test behavior and failing tests on master

passalis, 2021-07-07:
It seems that tests are randomly failing on `tools.perception.activity_recognition.cox3d.test_cox3d_learner.TestCoX3DLearner`. 
More specifically, the error message is:
```
FAIL: test_infer (tools.perception.activity_recognition.cox3d.test_cox3d_learner.TestCoX3DLearner)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/opendr_internal/opendr_internal/tests/sources/tools/perception/activity_recognition/cox3d/test_cox3d_learner.py", line 112, in test_infer
    assert torch.allclose(results1[0].confidence, results2[0].confidence)
AssertionError
```
I can also reproduce this behavior on the master branch when I run the tests locally. Examining different tests runs, e.g,. this [one](https://github.com/tasostefas/opendr_internal/pull/98/checks?check_run_id=3001859356) and a [failing one](https://github.com/tasostefas/opendr_internal/pull/99/checks?check_run_id=3003072825), it seems that the problematic test does not always run. I am not sure why in some cases the test runs stop early.  Note that in the first case 58 tests ran, while in the second one 66 tests ran.

So, I see two current issues:

1) A different number of tests run without a clear explanation on why this happens. 

2) The  `TestCoX3DLearner` fails for me locally on the master branch and it also seems to fail on other branches as well. So, I tend to believe that this test might passed without actually running. The fix seems for this seems to be quite easy. Just raising slightly the relative error threshold on *allclose()*, i.e., setting `atol=1e-07` from the default value of `atol=1e-08`, seems to solve the issue, but perhaps @LukasHedegaard wants to check a) if the problem indeed exists and b) if this is a valid solution.

LukasHedegaard, 2021-07-11:
Thank you for bringing this to my attention @passalis. 
I have not experienced the issue on the machines I have tested it on (MacOS 10 and Ubuntu 18), but on other occasions, I have observed small numerical difference between environments. I think your diagnosis and proposed solution (increasing `atol` in `allclose`) is an acceptable practical solution. 
I have created a PR #105 that implements the fix.

passalis, 2021-07-12:
Thanks @LukasHedegaard for fixing this!

passalis, 2021-08-30:
I am closing this, since this issue does not seem to affect testing any more.

closed 2021-08-30 by passalis