opendr-eu/opendr #110: Conflicting python dependencies, e.g., onnxruntime

vniclas, 2021-07-15:
Hi all,

when looking into the current dependency handling of python packages, I realized that there are conflicting dependencies between the integrated tools, e.g., `onnxruntime==1.5.2` vs `onnxruntime==1.3.0`.
The only reason that the tests have not failed until now is that parsing the python dependencies is done manually line by line (see [install.sh line 18](https://github.com/tasostefas/opendr_internal/blob/master/dependencies/install.sh), and hence silently overwriting previous package versions, instead of using a smart way of resolving conflicts.

Is a reason for the manual parsing instead of using `pip install -r python_dependencies.txt`?

stefaniapedrazzi, 2021-07-19:
> Is a reason for the manual parsing instead of using `pip install -r python_dependencies.txt`?

No, probably there is no particular reason other that the manual parsing is needed for the system dependencies and can probably be changed.

Please feel free to go ahead fixing this issue.
Otherwise I will handle it when I will have some time.

vniclas, 2021-07-20:
@stefaniapedrazzi, thanks for your reply. Please see PR #112 for the further discussion.

closed 2021-08-18 by vniclas