opendr-eu/opendr #114: Test failures in test_suite on master branch due to TestObjectTracking2DFairMotLearner

stefaniapedrazzi, 2021-07-22:
Currnetly on the master branch the `TestObjectTracking2DFairMotLearner` tests are failing.

Here is the error I got locally (on the CI the logs are cut because too long):
```
======================================================================
ERROR: test_eval (tools.perception.object_tracking_2d.fair_mot.test_object_tracking_2d_fair_mot.TestObjectTracking2DFairMotLearner)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/stefi/develop/opendr_internal/tests/sources/tools/perception/object_tracking_2d/fair_mot/test_object_tracking_2d_fair_mot.py", line 168, in test_eval
    test_model(name)
  File "/home/stefi/develop/opendr_internal/tests/sources/tools/perception/object_tracking_2d/fair_mot/test_object_tracking_2d_fair_mot.py", line 163, in test_model
    result = learner.eval(eval_dataset)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/object_tracking_2d_fair_mot_learner.py", line 346, in eval
    result = evaluate(self.infer, dataset)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/algorithm/run.py", line 144, in evaluate
    output_target = infer(image, frame_id)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/object_tracking_2d_fair_mot_learner.py", line 388, in infer
    online_targets = self.tracker.update(blob, img0)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/algorithm/lib/tracker/multitracker.py", line 276, in update
    output = self.model(im_blob)[-1]
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/algorithm/lib/models/networks/pose_dla_dcn.py", line 598, in forward
    x = self.dla_up(x)
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/algorithm/lib/models/networks/pose_dla_dcn.py", line 507, in forward
    ida(layers, len(layers) - i - 2, len(layers))
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/algorithm/lib/models/networks/pose_dla_dcn.py", line 479, in forward
    layers[i] = upsample(project(layers[i]))
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/stefi/develop/opendr_internal/src/opendr/perception/object_tracking_2d/fair_mot/algorithm/lib/models/networks/pose_dla_dcn.py", line 445, in forward
    x = self.conv(x)
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/DCN/dcn_v2.py", line 128, in forward
    self.deformable_groups)
  File "/home/stefi/develop/opendr_internal/venv/lib/python3.6/site-packages/DCN/dcn_v2.py", line 31, in forward
    ctx.deformable_groups)
RuntimeError: Not compiled with GPU support
```

@iliiliiliili Could you please have a look at this error?

stefaniapedrazzi, 2021-09-07:
This may also be related to #125.

We should then check and enable again this test.

passalis, 2021-12-16:
@iliiliiliili this is now resolved. Is this correct?

closed 2023-07-03 by stefaniapedrazzi