opendr-eu/opendr #119: Double requirements for pip

iliiliiliili, 2021-08-28:
Using `make install_compilation_dependencies` results in this error:

`ERROR: Double requirement given: tensorboard>=2.3.0 (from -r python_dependencies.txt (line 18)) (already in tensorboard==2.4.1 (from -r python_dependencies.txt (line 14)), name='tensorboard')`

stefaniapedrazzi, 2021-08-30:
In which branch do you have this error?
I just checked on the master branch and everything seems to work fine.

passalis, 2021-08-30:
I also encountered this error once. However, I think that this was due to some misconfiguration on my system. It was solved by cloning the repo again to be in a clean state and creating a new clean venv. I can no longer reproduce this, so I am not sure what caused this.

vniclas, 2021-09-20:
This should not be possible anymore with the latest changes on master (#112) but result in failed tests during CI.

Thus, this issue can probably be closed. @iliiliiliili?

passalis, 2021-09-21:
I am still facing this issue locally (on c_face_recognition branch, which is up-to-date with the master). The error is the same

```
ERROR: Double requirement given: tensorboard==2.4.1 (from -r python_dependencies.txt (line 36)) (already in tensorboard>=2.3.0 (from -r python_dependencies.txt (line 11)), name='tensorboard')
```

This is on a clean repo clone and on fresh Ubuntu 20.04 installation. It is very strange, since I am just running the same commands as the CI setup script.

passalis, 2021-09-21:
I found out that this originates also from face recognition (this tool uses a different version for matplotlib), so I will fix it within the same PR.

closed 2021-09-21 by passalis