opendr-eu/opendr #127: Tests stuck on activity recognition

passalis, 2021-09-21:
I think that tests now get stuck at activity recognition tests for no specific reason. I can successfully run this tests locally. However, there are three runs in which these tests seem to have stuck:

https://github.com/tasostefas/opendr_internal/pull/94/checks?check_run_id=3662854127
https://github.com/tasostefas/opendr_internal/pull/107/checks?check_run_id=3662837030
https://github.com/tasostefas/opendr_internal/pull/89/checks?check_run_id=3661815195

passalis, 2021-09-21:
@LukasHedegaard could please take a look on this?

It seems that the tests were also failed on the recent bugfix (https://github.com/tasostefas/opendr_internal/runs/3660218030) that was merged on the master branch.

LukasHedegaard, 2021-09-22:
My apologies, it seems I merged the branch prematurely. A PR #128 with a proposed fix has been opened.

passalis, 2021-09-23:
#128 resolved this issue. Thank you!

closed 2021-09-23 by passalis