opendr-eu/opendr #131: Consistency of input for various algorithms contained in OpenDR

passalis, 2021-10-05:
Different implementations might receive input in different formats in OpenDR toolkit. For example, the shape of the image input can be either *NHWC* or *NCHW*, where *N* is the number of images, *H/W* is the height/width of an image, and *C* denotes its channels. PyTorch defaults to *NCHW*, while TensorFlow defaults to *NHWC*. OpenDR should obviously select and stick with one choice to ensure the required homogeneity and transparency between different algorithms. Currently, (as far as I understand) there is no check performed to ensure that the image is supplied in the correct format. However, switching between different formats is easy, so existing implementations can be easily adjusted to account for this difference. 

You can find a discussion regarding the benefits of the one or the other [here](https://stackoverflow.com/questions/44280335/how-much-faster-is-nchw-compared-to-nhwc-in-tensorflow-cudnn). However, I think this is not relevant to our choice, since we are discussing the *interface* of the toolkit and not the actual implementation. Given that the two biggest toolkits (PyTorch and TensorFlow) have selected different formats, I think it is enough to select one and stick to our choice.

Apart from the issue regarding the *NCHW*/*NHWC*, we should also decide on the channel order (i.e., *BGR* vs *RGB*). Unfortunately, this is a much harder issue to detect and sometimes it can go unnoticed since models will correctly perform the feedfoward pass regardless of the used channel order, while in some cases they might even produce the quite good results (channel flip is actually a data augmentation technique!). So, a second choice that we should make is whether the default format will be RGB or BGR. Again, given that you know how the models have been trained, switching between the different formats is easy and we can easily ensure that all implementations receive the data as expected.

Therefore, we should:
[ ] Decide between NHWC or NCHW for images
[ ] Decide between RGB or BGR for images
[ ] Researchers working on Videos/Timeseries and/or other kinds of input provide their opinion on relevant issues for the input. 

My recommendation would be to have (N)CHW / BGR format for images, (N)TC format (T denotes time) for timeseries and  (N)TCHW (?) / BGR format for videos (if this makes sense, otherwise we should stick to the most common native format to avoid conversions and the corresponding impact on the performance).

LukasHedegaard, 2021-10-05:
I believe it would make sense to adopt one of the existing common standards. I do think that most perception algorithms in OpenDR use PyTorch. As far as I can see, the `torchvision` project uses channel order _RGB_ (see [line 251](https://github.com/pytorch/vision/blob/master/torchvision/datasets/folder.py#L42)), _NCHW_ for images and _NCTHW_ for videos (see [here](https://github.com/pytorch/vision/blob/master/torchvision/transforms/_transforms_video.py)).

In conclusion, I would vote use this standard, e.g. _RGB_, _NCHW_, and _NCTHW_.

vniclas, 2021-10-05:
I agree with @LukasHedegaard to follow the _torchvision_ conventions. I would further propose to implement `from_file()` functions for the various `Data` classes, e.g., to ensure that images are loaded as RGB.

vniclas, 2021-11-03:
How do we proceed to ensure that all of the implemented algorithms respect these conventions? It would probably make sense to discuss (or at least announce it) in one of the next meetings.

ad-daniel, 2021-12-03:
Please check if you have ensured your tool adheres to the convention in Nikos' email.

**Perception:**

- [ ] Face Detection
   - [ ] RetinaFace
- [ ] Face Recognition
   - [ ] Face recognition
- [ ] Landmark based facial expression recognition
   - [ ] Landmark based facial expression recognition
- [ ] Human Pose Estimation
   - [ ] Lightweight open pose
- [ ] Video Activity Recognition
   - [ ] x3d
- [ ] Online Video Activity Recognition
   - [ ] cox3d
- [ ] Skeleton based Action Recognition
   - [ ] spatio temporal GCN
- [ ] Progressive Skeleton based Action Recognition
   - [ ] progressive spation temporal GCN
- [ ] Speech recognition
   - [ ] quadraticselfonn
   - [ ] matchboxnet
   - [ ] edgesspeechnets
- [ ] Heart Anomaly Detection 
   - [ ] Neural bag-of-feature
   - [ ] Gated Recurrent Unit
- [ ] Semantic Scene segmentation 
   - [ ] BiseNet
- [ ] Object detection 2d
   - [ ] Centernet
   - [ ] SSD
   - [ ] Yolov3
- [ ] Efficient Panoptic Segmentation
   - [ ] Panoptic segmentation
- [ ] 3d Object Localization
   - [ ] Voxel object detection 3d
- [ ] Object Tracking 2d/3d
   - [ ] Deep sort
   - [ ] Fair mot
   - [ ] ab3dmot
- [ ] Multimodal Human Centric Perception
  - [ ] RGBD hand gesture
- [ ] Multimodal 2d object detection
  - [ ] GEM
  - [ ] Detr

**Control:**
- [ ] Mobile manipulation
- [ ] Single Demostration Grasping

**Planning:**
- [ ] End-to-End planner

**Simulation:**
- [ ] Human Model Generation

passalis, 2021-12-06:
I have just realized that `Image.open(image_path)` does not follow NCHW conventions. Therefore, most probably all existing tools already work in NHWC instead of NCHW (they already change the channel order internally).

LukasHedegaard, 2021-12-06:
Frankly, I did not test this functionality in the tools, I integrated. The tested images were loaded in another manner without the need for dimension transposition.
So changing `Image.open(image_path)` to use the agreed upon NCHW format, would actually remove a potential bug in my case 😅

vniclas, 2021-12-06:
This might actually be a bit more confusing than we initially thought. The `to_tensor()` (torchvision), converting PIL images or NumPy arrays to a tensor, expects an HWC input and outputs CHW as expected by PyTorch.
Thus, if we load images as CHW, we have to make people aware that they have to cast back to HWC before using the built-in `to_tensor()` function in their dataloaders.

vniclas, 2021-12-06:
Just to repeat what I also said during the TMT Meeting, we'll have to adjust the [OpenDR](https://github.com/tasostefas/opendr_internal/blob/master/projects/opendr_ws/src/ros_bridge/src/opendr_bridge/bridge.py) bridge as well since OpenCV assumes HWC.

passalis, 2021-12-07:
I have opened #164 according to the discussion we had during the last meeting. Please note that I also added an opencv() method to easily obtaining the images into a format that can be used for visualization (check the already updated examples to see its usage). I will also update the documentation and update the rosbridge asap.

vniclas, 2021-12-09:
Then we can probably close this issue now.

closed 2021-12-09 by passalis