opendr-eu/opendr #132: Simplify the import path of algorithms

passalis, 2021-10-05:
Currently, implementations do not expose the main *Learner* class, as well as any of the developed tools in the root **__init__.py **.  This results to very long import paths that negatively affects the ease of use of the toolkit. For example, my suggestion would be to go from:

```python
from opendr.perception.pose_estimation.lightweight_open_pose.lightweight_open_pose_learner import \
    LightweightOpenPoseLearner
```

to simply

```python
from opendr.perception.pose_estimation import LightweightOpenPoseLearner
```

This can be easily achieved simply by appropriately importing the Learner class in the algorithm root's (e.g., `opendr.perception.pose_estimation`) `__init__.py` file.

I will open a PR soon demonstrating the how this can be achieved.

vniclas, 2021-11-23:
@ad-daniel This should probably also be one of the ToDos before the release.

ad-daniel, 2021-11-23:
Yes, can probably be done at the same time/after when we update the base classes.

passalis, 2021-12-03:
I think this issue has been addressed in #133. I have also updated the contribution instructions in the wiki (https://github.com/tasostefas/opendr_internal/wiki/Adding-New-Tool).

closed 2021-12-03 by passalis