opendr-eu/opendr #148: Status of ROS interface

ad-daniel, 2021-11-01:
**Perception:**

- [x] Face Detection
   - [x] RetinaFace
- [ ] Face Recognition
   - [ ] Face recognition https://github.com/tasostefas/opendr_internal/pull/101
- [ ] Landmark based facial expression recognition
   - [ ] Landmark based facial expression recognition
- [x] Human Pose Estimation
   - [x] Lightweight open pose
- [ ] Video Activity Recognition
   - [ ] x3d
- [ ] Online Video Activity Recognition
   - [ ] cox3d
- [ ] Skeleton based Action Recognition
   - [ ] spatio temporal GCN
- [ ] Progressive Skeleton based Action Recognition
   - [ ] progressive spation temporal GCN
- [ ] Speech recognition
   - [ ] quadraticselfonn
   - [ ] matchboxnet
   - [ ] edgesspeechnets
- [ ] Heart Anomaly Detection
   - [ ] Neural bag-of-feature
   - [ ] Gated Recurrent Unit
- [ ] Semantic Scene segmentation
   - [ ] BiseNet
- [x] Object detection 2d
   - [x] Centernet
   - [x] SSD
   - [x] Yolov3
- [x] Efficient Panoptic Segmentation
   - [x] Panoptic segmentation
- [ ] 3d Object Localization
   - [ ] Voxel object detection 3d
- [ ] Object Tracking 2d/3d
   - [ ] Deep sort
   - [ ] Fair mot
   - [ ] ab3dmot
- [ ] Multimodal Human Centric Perception
  - [ ] RGBD hand gesture
- [x] Multimodal 2d object detection
  - [x] GEM
  - [x] Detr

**Control:**
- [ ] mobile manipulation
- [ ] Single Demostration Grasping

**Planning:**
- [ ] End-to-End planner

**Simulation:**
- [ ] Human Model Generation

passalis, 2021-12-15:
I updated the list with recent PRs adding more nodes. I checked the tasks for which there is an open PR, in order to keep track what is actually missing.

passalis, 2021-12-23:
I see that we are no longer missing any ROS node. I am closing this issue for now.

closed 2021-12-23 by passalis