opendr-eu/opendr #169: Missing/Incomplete Demos

passalis, 2021-12-13:
I have identified at least the following tools/tasks for which we do not provide any demo in projects (or we might need a few tweaks - please:

**Perception:**
- [ ] Speech recognition @jovaroo 
   - [ ] quadraticselfonn
   - [ ] matchboxnet
   - [ ] edgesspeechnets
- [ ] Heart Anomaly Detection  @viebboy 
   - [ ] Neural bag-of-feature
   - [ ] Gated Recurrent Unit
- [ ] Semantic Scene segmentation @mtzelepi 
   - [ ] BiseNet
- [ ] Multimodal Human Centric Perception @viebboy 
  - [ ] RGBD hand gesture 

**Control:**
- [ ] Mobile manipulation (demo exists, but missing a small readme like the rest) @dHonerkamp 

Demos for tools not yet integrated
- [ ] Single Demostration Grasping 

**Planning:**
- [ ] End-to-End planner

dHonerkamp, 2021-12-13:
Added a PR for the mobile manipulation demo: #170

sisaha9, 2022-01-26:
@passalis What are the current considerations for the end to end planner?

passalis, 2022-01-26:
There is a PR open #82 for this tool.

passalis, 2022-02-10:
I am closing this, since there all demos have been integrated in the first version, apart from the end-to-end planner that will be part of the next release.

closed 2022-02-10 by passalis