opendr-eu/opendr #190: WSL support

iliiliiliili, 2021-12-20:
Currently, [WSL](https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux) is not fully supported.
The CPU wheel installation works on Ubuntu 18, but fails on WSL1 Ubuntu 18 for Windows 10 with the following error:
 ```
  Using cached matplotlib-2.2.2.tar.gz (37.3 MB)
    ERROR: Command errored out with exit status 1:
     command: /home/io/miniconda3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-68_0e5d6/matplotlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-68_0e5d6/matplotlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-68_0e5d6/matplotlib/pip-egg-info
         cwd: /tmp/pip-install-68_0e5d6/matplotlib/
    Complete output (64 lines):
    IMPORTANT WARNING:
        pkg-config is not installed.
        matplotlib may not be able to find some of its dependencies
    ============================================================================
    Edit setup.cfg to change the build options

    BUILDING MATPLOTLIB
                matplotlib: yes [2.2.2]
                    python: yes [3.8.3 (default, May 19 2020, 18:47:26)  [GCC
                            7.3.0]]
                  platform: yes [linux]

    REQUIRED DEPENDENCIES AND EXTENSIONS
                     numpy: yes [version 1.21.0]
          install_requires: yes [handled by setuptools]
                    libagg: yes [pkg-config information for 'libagg' could not
                            be found. Using local copy.]
                  freetype: no  [The C/C++ header for freetype2 (ft2build.h)
                            could not be found.  You may need to install the
                            development package.]
                       png: yes [version 1.6.34]
                     qhull: yes [pkg-config information for 'libqhull' could not
                            be found. Using local copy.]

    OPTIONAL SUBPACKAGES
               sample_data: yes [installing]
                  toolkits: yes [installing]
                     tests: no  [skipping due to configuration]
            toolkits_tests: no  [skipping due to configuration]

    OPTIONAL BACKEND EXTENSIONS
                    macosx: no  [Mac OS-X only]
                    qt5agg: no  [PySide2 not found; PyQt5 not found]
                    qt4agg: no  [PySide not found; PyQt4 not found]
                   gtk3agg: no  [Requires pygobject to be installed.]
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-68_0e5d6/matplotlib/setup.py", line 197, in <module>
        msg = pkg.install_help_msg()
      File "/tmp/pip-install-68_0e5d6/matplotlib/setupext.py", line 592, in install_help_msg
        release = platform.linux_distribution()[0].lower()
    AttributeError: module 'platform' has no attribute 'linux_distribution'
                 gtk3cairo: no  [Requires cairocffi or pycairo to be installed.]
                    gtkagg: no  [Requires pygtk]
                     tkagg: yes [installing; run-time loading from Python Tcl /
                            Tk]
                     wxagg: no  [requires wxPython]
                       gtk: no  [Requires pygtk]
                       agg: yes [installing]
                     cairo: no  [cairocffi or pycairo not found]
                 windowing: no  [Microsoft Windows only]

    OPTIONAL LATEX DEPENDENCIES
                    dvipng: no
               ghostscript: no
                     latex: no
                   pdftops: no

    OPTIONAL PACKAGE DATA
                      dlls: no  [skipping due to configuration]

    ============================================================================
                            * The following required packages can not be built:
                            * freetype
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
The installation of `libfreetype6-dev` by 'sudo apt-get install --yes libfreetype6-dev' does not change the result.

iliiliiliili, 2021-12-20:
The same happens in WSL1 Ubuntu 20 for Windows 10:
```
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-5162521x/matplotlib_e75f2f4575ef44bfacfdf5bcdd907aac/setup.py'"'"'; __file__='"'"'/tmp/pip-install-5162521x/matplotlib_e75f2f4575ef44bfacfdf5bcdd907aac/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-zcl5yc0k
       cwd: /tmp/pip-install-5162521x/matplotlib_e75f2f4575ef44bfacfdf5bcdd907aac/
  Complete output (65 lines):
  ============================================================================
  Edit setup.cfg to change the build options

  BUILDING MATPLOTLIB
              matplotlib: yes [2.2.2]
                  python: yes [3.8.10 (default, Sep 28 2021, 16:10:42)  [GCC
                          9.3.0]]
                platform: yes [linux]

  REQUIRED DEPENDENCIES AND EXTENSIONS
                   numpy: yes [version 1.17.4]
        install_requires: yes [handled by setuptools]
                  libagg: yes [pkg-config information for 'libagg' could not
                          be found. Using local copy.]
                freetype: no  [The C/C++ header for freetype2 (ft2build.h)
                          could not be found.  You may need to install the
                          development package.]
                     png: yes [version 1.6.37]
                   qhull: yes [pkg-config information for 'libqhull' could not
                          be found. Using local copy.]

  OPTIONAL SUBPACKAGES
             sample_data: yes [installing]
                toolkits: yes [installing]
                   tests: no  [skipping due to configuration]
          toolkits_tests: no  [skipping due to configuration]

  OPTIONAL BACKEND EXTENSIONS
                  macosx: no  [Mac OS-X only]
                  qt5agg: no  [PySide2 not found; PyQt5 not found]
                  qt4agg: no  [PySide not found; PyQt4 not found]
  Unable to init server: Could not connect: Connection refused
  Unable to init server: Could not connect: Connection refused
                 gtk3agg: yes [installing, version 3.20.24]
  Unable to init server: Could not connect: Connection refused
  Unable to init server: Could not connect: Connection refused
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-5162521x/matplotlib_e75f2f4575ef44bfacfdf5bcdd907aac/setup.py", line 197, in <module>
      msg = pkg.install_help_msg()
    File "/tmp/pip-install-5162521x/matplotlib_e75f2f4575ef44bfacfdf5bcdd907aac/setupext.py", line 592, in install_help_msg
      release = platform.linux_distribution()[0].lower()
  AttributeError: module 'platform' has no attribute 'linux_distribution'
               gtk3cairo: yes [installing, version 3.20.24]
                  gtkagg: no  [Requires pygtk]
                   tkagg: yes [installing; run-time loading from Python Tcl /
                          Tk]
                   wxagg: no  [requires wxPython]
                     gtk: no  [Requires pygtk]
                     agg: yes [installing]
                   cairo: yes [installing, pycairo version 1.16.2]
               windowing: no  [Microsoft Windows only]

  OPTIONAL LATEX DEPENDENCIES
                  dvipng: no
             ghostscript: no
                   latex: no
                 pdftops: no

  OPTIONAL PACKAGE DATA
                    dlls: no  [skipping due to configuration]

  ============================================================================
                          * The following required packages can not be built:
                          * freetype
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/ec/ed/46b835da53b7ed05bd4c6cae293f13ec26e877d2e490a53a709915a9dcb7/matplotlib-2.2.2.tar.gz#sha256=4dc7ef528aad21f22be85e95725234c5178c0f938e2228ca76640e5e84d8cde8 (from https://pypi.org/simple/matplotlib/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement matplotlib==2.2.2 (from opendr) (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1)
ERROR: No matching distribution found for matplotlib==2.2.2
```

pkraison, 2022-01-05:
> The same happens in WSL1 Ubuntu 20 for Windows 10:
>
> ```
>   ERROR: Command errored out with exit status 1:
>    command: /usr/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = [...]

Hey, thanks for the amazing work. did you try just matplotlib in the requirements file, without version(2.2.2)

passalis, 2022-01-06:
We should try this, since this version can create issues also on recent Linux systems and might easily solve the issues we have.

I have also tried to run the CPU docker on windows and seems to run without any issues (Windows 10 - Build 19043.1415, Docker Desktop  4.3.2). So, it seems that we have at this point at least one way to make the toolkit accessible to Windows users without any major issue (as far as I tested it). I guess that it will also work on Mac OS. I will try asap.

passalis, 2022-08-31:
I am closing this issue, since we are explicitly referring to unsupported system configurations in #293

closed 2022-08-31 by passalis