opendr-eu/opendr #194: Error while installing with pip

gamzeakyol, 2022-01-06:
Hi,

I would like to install opendr with pip as instructed in installation.md but had an error. Error message says:

ERROR: Could not find a version that satisfies the requirement opendr-toolkit (from versions: none)
ERROR: No matching distribution found for opendr-toolkit

Could you please help for this situation?

Thanks,

ad-daniel, 2022-01-06:
Hi, I've tried again and it appears to be working for me (Ubuntu 20). Are you using Ubuntu 20? Can you provide the full output when executing the instructions, in case something else isn't working.

passalis, 2022-01-06:
As @ad-daniel suggested, we need some additional information to troubleshoot this. 

I suspect that this is most probably related to not using Python 3.8 (currently a pre-built wheel is only provided for 3.8, since toolkit has been tested only on Python 3.8). There is no sdist or bdist wheel for other distributions so pip will not find a match for these.

If this is the case (e.g., if you are using Python 3.6/3.7), then pip installation is not currently supported. I will try to prepare a package for other Python versions as well, to resolve this issue, since the toolkit indeed seems to run on these without problem (however, this is still an untested configuration, so you should proceed with caution). 

In any case, I strongly advise to first try the toolkit within the docker containers that seem to work without any issue on various platforms (even on Windows).

passalis, 2022-02-03:
We have already switched into using source distributions, so this should be working (however, I am not sure if all dependencies can be met). We are also working on providing modular packages that will have a smaller number of dependencies and will be easier to install (#201)

passalis, 2022-06-15:
We have release the new version that has switched into using source distributions. This issue should be now resolved. @gamzeakyol would be possible to kindly check if this solves the issue for you?

passalis, 2022-07-11:
I am closing this issue, since it seems that this has been addressed in the new release. We can re-open it if the problem comes back.

closed 2022-07-11 by passalis