opendr-eu/opendr #205: [Face_recognition] Miss explanation of adding costumed database + bug

thomaspeyrucain, 2022-01-27:
```
(venv) thomaspeyrucain@manama:~/Desktop/OpenDR/opendr/projects/opendr_ws$ rosrun perception face_recognition.py 
GPU found.
Model already exists
Loading backbone 'mobilefacenet'
Loading Reference
[ERROR] [1643301733.091029]: bad callback: <bound method FaceRecognitionNode.callback of <__main__.FaceRecognitionNode object at 0x7fa74de53f70>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/Desktop/OpenDR/opendr/projects/opendr_ws/src/perception/scripts/face_recognition.py", line 90, in callback
    self.face_publisher.publish(ros_face)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 878, in publish
    raise ROSException("ROS node has not been initialized yet. Please call init_node() first")
rospy.exceptions.ROSException: ROS node has not been initialized yet. Please call init_node() first

[ERROR] [1643301733.115256]: bad callback: <bound method FaceRecognitionNode.callback of <__main__.FaceRecognitionNode object at 0x7fa74de53f70>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/Desktop/OpenDR/opendr/projects/opendr_ws/src/perception/scripts/face_recognition.py", line 90, in callback
    self.face_publisher.publish(ros_face)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 878, in publish
    raise ROSException("ROS node has not been initialized yet. Please call init_node() first")
rospy.exceptions.ROSException: ROS node has not been initialized yet. Please call init_node() first

[INFO] [1643301733.139685]: Face recognition node started!

```
It says GPU found but I am with cuda 11.2 it should detect that it is not compatible,
I get this error when running the script

passalis, 2022-01-31:
Torch bundles its own cuda distribution, so this should work regardless the cuda that you have installed on your system (so if it says that a GPU is detected, most probably it will work).

I think that issue here is that the node subscribes to a topic before being initialized, which probably leads to these errors. However, I think that the node should still work. @Pavlos-Tosidis can you please check this?

Also, in the code it seems that this node receives the input directly from the camera. Isn't this node supposed to directly receive the cropped bounding boxes?

Pavlos-Tosidis, 2022-02-09:
These are addressed in #219

passalis, 2022-03-09:
I see that #219 has been merged. @thomaspeyrucain @Pavlos-Tosidis is this resolved? should we close the issue?

Pavlos-Tosidis, 2022-03-09:
This is resolved, waiting for dev branch to be merged into master to close this.

closed 2022-03-09 by passalis