opendr-eu/opendr #210: Git clone installation on Ubuntu

iliiliiliili, 2022-02-02:
After clonning the repository on a new Ubuntu 18 user, and running `./bin/install.sh`, it failed after some time with `face_recognition.cpp:23:10: fatal error: filesystem: No such file or directory`:

```
g++ -pthread -shared -B /opt/anaconda/compiler_compat -L/opt/anaconda/lib -Wl,-rpath=/opt/anaconda/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/nms_kernel.o build/temp.linux-x86_64-3.7/gpu_nms.o -L/usr/local/cuda/lib64 -Wl,-R/usr/local/cuda/lib64 -lcudart -o /home/io1/opendr/src/opendr/perception/object_detection_2d/retinaface/algorithm/cython/gpu_nms.cpython-37m-x86_64-linux-gnu.so
make[1]: Leaving directory '/home/io1/opendr/src/opendr/perception/object_detection_2d/retinaface'
make[1]: Entering directory '/home/io1/opendr/src/c_api'
Building C API...
g++ -fPIC -c opendr_utils.cpp -o /home/io1/opendr/build/opendr_utils.o -I/usr/local/include/onnxruntime/ `pkg-config --cflags opencv4` -I/home/io1/opendr/include
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
g++ -fPIC -c face_recognition.cpp -o /home/io1/opendr/build/opendr_face_recognition.o -I/usr/local/include/onnxruntime/ `pkg-config --cflags opencv4` -I/home/io1/opendr/include
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
face_recognition.cpp:23:10: fatal error: filesystem: No such file or directory
 #include <filesystem>
          ^~~~~~~~~~~~
compilation terminated.
Makefile:38: recipe for target '/home/io1/opendr/lib/libopendr.so' failed
make[1]: *** [/home/io1/opendr/lib/libopendr.so] Error 1
make[1]: Leaving directory '/home/io1/opendr/src/c_api'
Makefile:55: recipe for target 'libopendr' failed
make: *** [libopendr] Error 2
```

iliiliiliili, 2022-02-02:
It looks like this problem does not exist on WSL2 + Ubuntu 20

passalis, 2022-08-31:
I am closing this issue, since we are now explicitly referring to unsupported system configurations in #293

closed 2022-08-31 by passalis