opendr-eu/opendr #216: onnxruntime installation error

negarhdr, 2022-02-08:
After cloning the repository on TX2 and Xavier systems with Ubuntu 18 , and running ./bin/install.sh, it failed after some time with the following error:

> Building C API...
g++ -fPIC -c opendr_utils.cpp -o /home/maleci/opendr/build/opendr_utils.o -I/usr/local/include/onnxruntime/ `pkg-config --cflags opencv4` -I/home/maleci/opendr/include
g++ -fPIC -c face_recognition.cpp -o /home/maleci/opendr/build/opendr_face_recognition.o -I/usr/local/include/onnxruntime/ `pkg-config --cflags opencv4` -I/home/maleci/opendr/include
g++ -fPIC  -o /home/maleci/opendr/lib/libopendr.so /home/maleci/opendr/build/opendr_utils.o /home/maleci/opendr/build/opendr_face_recognition.o -L/usr/lib/ -L/usr/local/lib/ -lstdc++ -lm -lonnxruntime `pkg-config --libs opencv4` -lboost_filesystem  --shared
/usr/bin/ld: skipping incompatible /usr/local/lib//libonnxruntime.so when searching for -lonnxruntime
/usr/bin/ld: skipping incompatible //usr/local/lib/libonnxruntime.so when searching for -lonnxruntime
/usr/bin/ld: cannot find -lonnxruntime
collect2: error: ld returned 1 exit status
Makefile:38: recipe for target '/home/maleci/opendr/lib/libopendr.so' failed
make[1]: *** [/home/maleci/opendr/lib/libopendr.so] Error 1
make[1]: Leaving directory '/home/maleci/opendr/src/c_api'
Makefile:55: recipe for target 'libopendr' failed
make: *** [libopendr] Error 2
./bin/install.sh: line 43: deactivate: command not found

iliiliiliili, 2022-02-08:
Installation of the Jetson onnx runtime from the [Nvidia Jetson Onnx Runtime](https://developer.nvidia.com/blog/announcing-onnx-runtime-for-jetson/) did not fix the problem

passalis, 2022-02-09:
Please use the [install_tx2.sh](https://github.com/opendr-eu/opendr/blob/tx2_install/bin/install_tx2.sh) script for installing in TX-2. The error log shows that the installation fails when compiling the C API against ONNX (which is expected, since `install.sh` downloads an x86 pre-compiled version of onnx). If the pip wheel that is available for jetson contains the headers, then we can point the compiler there and I expect that it will work. Otherwise, we might need to compile ONNX to enable the C API on jetsons. However, note that this is only for the C API. Using the [install_tx2.sh](https://github.com/opendr-eu/opendr/blob/tx2_install/bin/install_tx2.sh) should be enough to enable the Python API of the toolkit.

Pavlos-Tosidis, 2022-03-11:
Closing this as for supported Nvidia's embedded devices, the installation script @passalis mentioned should be used to install the toolkit.

closed 2022-03-11 by Pavlos-Tosidis