opendr-eu/opendr #218: Pytorch installation on Jetson

iliiliiliili, 2022-02-08:
Pytorch for Jetson devices with GPU support cannot be installed directly and has to be installed using [provided pip wheels](https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-10-now-available/72048). These wheels are available for Python 3.6 only and cannot be installed via pip install torch==1.7.1. How should it be incorporated into Jetson installation? I believe, it is best to have separate install scripts for PC and Jetson devices.

passalis, 2022-02-09:
Did you use the [jetson_install.sh](https://github.com/opendr-eu/opendr/blob/tx2_install/bin/install_tx2.sh) script? I think it downloads one of these wheels.

vniclas, 2022-02-10:
@passalis I am a bit confused concerning the installation on Jetson devices. Is the script linked in your reply supposed to be used for all Jetson models or just for TX2?

passalis, 2022-02-10:
The script (I fixed the name in my previous answer) was developed for TX-2. Also, I am not sure that it supports all the tools, so each partner should check its own tools (#195 is open). It can be also used as a template to develop similar scripts for other jetsons.

Pavlos-Tosidis, 2022-02-12:
There are instructions in #195 on how to run the script on Xavier AGX and NX. As @passalis mentioned, some tools are not supported due to some dependencies missing, so each partner should check its own tools.

Pavlos-Tosidis, 2022-03-11:
Closing this, as #195 address this.

closed 2022-03-11 by Pavlos-Tosidis