opendr-eu/opendr #220: Opencv-Python and online activity recognition

Pavlos-Tosidis, 2022-02-10:
Installation of the toolkit with git clone and ./bin/install.sh installs opencv-python-4.5.1.48. [Online activity recognition](https://github.com/opendr-eu/opendr/tree/master/projects/perception/activity_recognition/demos/online_recognition) won't work with this version. Installing opencv-python-4.5.5.62 fixes the problem.

LukasHedegaard, 2022-02-14:
You link to the Online activity recognition demo; is this the demo that doesn't work or do you experience issues with the activity recognition tool itself? 

If it is only the demo that has issues with `opencv-python-4.5.1.48`, we could note `opencv-python==4.5.5.62` in the associated [requirements.txt](https://github.com/opendr-eu/opendr/blob/master/projects/perception/activity_recognition/demos/online_recognition/requirements.txt) as a solution?

Pavlos-Tosidis, 2022-02-24:
I think we should run the tests with opencv-python==4.5.5.62 and if everything passes, change the requirements accordingly.

passalis, 2023-11-30:
Closing as outdated. If issue comes back, we can re-open.

closed 2023-11-30 by passalis