opendr-eu/opendr #225: Installation error on Ubuntu 18 with GPU support (face_recognition build, opencv2 missing)

LukasHedegaard, 2022-02-14:
## Steps to reproduce:
Follow [installation instructions](https://github.com/opendr-eu/opendr/blob/master/docs/reference/installation.md)
1. Clone toolkit
    ```
    git clone --depth 1 --recurse-submodules -j8 https://github.com/opendr-eu/opendr
    ```
1. Install toolkit
    ```
    cd opendr
    OPENDR_DEVICE=gpu ./bin/install.sh
    ```

## Error message:
```
...
g++ -fPIC -c face_recognition.cpp -o /home/lh/projects/opendr/build/opendr_face_recognition.o -I/usr/local/include/onnxruntime/ `pkg-config --cflags opencv4` -I/home/lh/projects/opendr/include
Package opencv4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv4' found
face_recognition.cpp:30:10: fatal error: opencv2/dnn/dnn.hpp: No such file or directory
 #include <opencv2/dnn/dnn.hpp>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:38: recipe for target '/home/lh/projects/opendr/lib/libopendr.so' failed
make[1]: *** [/home/lh/projects/opendr/lib/libopendr.so] Error 1
make[1]: Leaving directory '/home/lh/projects/opendr/src/c_api'
Makefile:55: recipe for target 'libopendr' failed
make: *** [libopendr] Error 2
```

## Discussion
The error message hints to a lacking co-installation of `opencv`. If it is assumed that `opencv` is installed prior to installing `opendr`, this should be noted in the [instructions](https://github.com/opendr-eu/opendr/blob/master/docs/reference/installation.md).
Otherwise, the build scripts should be fixed accordingly.

passalis, 2022-02-14:
The issue probably arises either due to opencv not being installed or `pkg-config` using a package name that is different on Ubuntu 18.04.  This issue seems to arise only when installing on an unsupported configuration (Ubuntu 18.04 is [not supported](https://github.com/opendr-eu/opendr/blob/master/docs/reference/installation.md) for clone&install), since it works as expected on Ubuntu 20.04 (see also #210).

 However, I think that the rest of the toolkit (apart from the C API) will most probably work on Ubuntu 18.04.

LukasHedegaard, 2022-02-15:
The rest of the toolkit does indeed work on Ubuntu 18. I had missed the lack of Ubuntu 18 support and the issue #210. 
Closing this issue as it is a duplicate  of #210.

closed 2022-02-15 by LukasHedegaard