opendr-eu/opendr #259: Skeleton-based Human Action Recognition Demo FAILS

davelbit, 2022-06-01:
After cloning the repository I was keen on testing the skeleton based action recognition, by issuing the following command:
`python3 demo.py --device cuda --video 0 --method stgcn --action_checkpoint_name stgcn_ntu_cv_lw_openpose` as suggested here: 
`https://github.com/opendr-eu/opendr/blob/master/projects/perception/skeleton_based_action_recognition/REAMDE.md`

But the program ended with the following error message:
`Traceback (most recent call last):
  File "demo.py", line 209, in <module>
    category_labels = preds2label(prediction.confidence)
  File "demo.py", line 112, in preds2label
    class_scores, class_inds = torch.topk(confidence, k=k)
TypeError: topk(): argument 'input' (position 1) must be Tensor, not float`

passalis, 2022-06-02:
Hi davelbit,

Thanks for your interest in our toolkit and for pointing out this bug. We replicated this bug and we are looking into this. We will get back soon.

passalis, 2022-06-03:
We have identified the cause  and fixed this issue in #260. Could you try the [HAR_demo_fix](https://github.com/opendr-eu/opendr/tree/HAR_demo_fix) branch to see if this fixes the issue for you?

davelbit, 2022-06-03:
I have cloned the HAR_demo_fix branch as suggested but found that the error still seems to exist. Can you confirm that the error is still present on your end?

passalis, 2022-06-04:
The bug related to `TypeError` seems to have been resolved for me. Please make sure that you source the new environment after cloning the branch (e.g., make sure that `PYTHONPATH` is pointing to the corrected clone). Let us know if this fixes the issue.

davelbit, 2022-06-05:
I missed to source the environment. It works now for me as well.

closed 2022-06-05 by davelbit