opendr-eu/opendr #299: rosnode - object_detection_2d_yolov3.py - error when echoing objects topic

thomaspeyrucain, 2022-09-12:
The output image is working but I cannot echo the object topic, I am using opendr installed on my computer on the develop branch

First I get this warning:
```
/home/user/opendr/venv/lib/python3.8/site-packages/numpy/core/fromnumeric.py:3432: RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
/home/user/opendr/venv/lib/python3.8/site-packages/numpy/core/_methods.py:190: RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
/home/user/opendr/src/opendr/perception/object_detection_2d/datasets/transforms.py:34: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  bbox_np = np.asarray([bbox.left, bbox.top, bbox.left + bbox.width, bbox.top + bbox.height, bbox.confidence, bbox.name])
```
And then I get this error when trying to echo the /open/objects:
`rostopic echo /open/objects`

```
[ERROR] [1662986163.583257]: bad callback: <bound method ObjectDetectionYOLONode.callback of <__main__.ObjectDetectionYOLONode object at 0x7f274299fc10>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/vision_msgs/msg/_Detection2DArray.py", line 247, in serialize
    buff.write(_get_struct_qd().pack(_x.id, _x.score))
TypeError: only integer scalar arrays can be converted to a scalar index

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  File "/opt/ros/noetic/lib/python3/dist-packages/vision_msgs/msg/_Detection2DArray.py", line 294, in serialize
    except TypeError as te: self._check_types(ValueError("%s: '%s' when writing '%s'" % (type(te), str(te), str(locals().get('_x', self)))))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 393, in _check_types
    check_type(n, t, getattr(self, n))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 314, in check_type
    check_type(field_name + '[]', base_type, v)
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 324, in check_type
    check_type('%s.%s' % (field_name, n), t, getattr(field_val, n))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 314, in check_type
    check_type(field_name + '[]', base_type, v)
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 324, in check_type
    check_type('%s.%s' % (field_name, n), t, getattr(field_val, n))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 261, in check_type
    raise SerializationError('field %s must be an integer type' % field_name)
genpy.message.SerializationError: field detections[].results[].id must be an integer type

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/user/opendr/projects/opendr_ws/src/perception/scripts/object_detection_2d_yolov3.py", line 95, in callback
    self.object_publisher.publish(ros_boxes)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: field detections[].results[].id must be an integer type

[ERROR] [1662986163.943623]: bad callback: <bound method ObjectDetectionYOLONode.callback of <__main__.ObjectDetectionYOLONode object at 0x7f274299fc10>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/vision_msgs/msg/_Detection2DArray.py", line 247, in serialize
    buff.write(_get_struct_qd().pack(_x.id, _x.score))
TypeError: only integer scalar arrays can be converted to a scalar index

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
[... 46 lines cut ...]
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 393, in _check_types
    check_type(n, t, getattr(self, n))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 314, in check_type
    check_type(field_name + '[]', base_type, v)
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 324, in check_type
    check_type('%s.%s' % (field_name, n), t, getattr(field_val, n))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 314, in check_type
    check_type(field_name + '[]', base_type, v)
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 324, in check_type
    check_type('%s.%s' % (field_name, n), t, getattr(field_val, n))
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 261, in check_type
    raise SerializationError('field %s must be an integer type' % field_name)
genpy.message.SerializationError: field detections[].results[].id must be an integer type

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/user/opendr/projects/opendr_ws/src/perception/scripts/object_detection_2d_yolov3.py", line 95, in callback
    self.object_publisher.publish(ros_boxes)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: field detections[].results[].id must be an integer type
```

tsampazk, 2022-09-13:
Hey @thomaspeyrucain this should be fixed in #300, feel free to test it and close the issue.

Warnings can be safely ignored, output gets produced properly.

thomaspeyrucain, 2022-09-13:
Hello again @tsampazk ,
Perfect thanks
I have just pulled and tested and it worked
I will wait for your ROS1 rosnode fix list before opening other issues so that I am sure it is not yet addressed in one of the MR

closed 2022-09-13 by thomaspeyrucain

tsampazk, 2022-09-13:
@thomaspeyrucain i just opened the tracker issue #305 to keep track of all updates.

PR #300 (2022-09-13): Fix ROS1 bridge type conversion bug
Closes #299 

Also added some ifs to stop re-downloading existing files in yolov3 object detection learner download method .