opendr-eu/opendr #301: rosnode - semantic_segmentation_bisenet.py - not working anymore

thomaspeyrucain, 2022-09-12:
The output image was working before but not anymore, I am using opendr installed on my computer on the develop branch and I got this warning and no image or topic are outputted when launching the node:
```
~/opendr/projects/opendr_ws/src/perception/scripts$ rosrun perception semantic_segmentation_bisenet.py 
Downloading: "https://download.pytorch.org/models/resnet18-f37072fd.pth" to /home/thomaspeyrucain/.cache/torch/hub/checkpoints/resnet18-f37072fd.pth
100%|██████████████████████████████████████████████████████████████| 44.7M/44.7M [00:06<00:00, 7.39MB/s]
Downloading: "https://download.pytorch.org/models/resnet101-63fe2227.pth" to /home/thomaspeyrucain/.cache/torch/hub/checkpoints/resnet101-63fe2227.pth
100%|████████████████████████████████████████████████████████████████| 171M/171M [00:22<00:00, 7.80MB/s]
Downloading pretrained weights...
Downloading json file...
[INFO] [1662972210.397098]: Semantic segmentation BiSeNet node started.
[WARN] [1662972210.524818]: Failed to generate prediction.
[WARN] [1662972210.586882]: Failed to generate prediction.
[WARN] [1662972210.652051]: Failed to generate prediction.
```

tsampazk, 2022-09-13:
Hey @thomaspeyrucain, i can't reproduce the issue locally unfortunately, as it runs properly. 

Could you please remove the try/except [here](https://github.com/opendr-eu/opendr/blob/364f0006560ae1089b50b4a90118ac557b8b5658/projects/opendr_ws/src/perception/scripts/semantic_segmentation_bisenet.py#L89-L111) and rerun the node? It should give you a proper exception message so we can make sense of what is happening.

thomaspeyrucain, 2022-09-13:
I get this error:
```
[ERROR] [1663054566.479403]: bad callback: <bound method BisenetNode.callback of <__main__.BisenetNode object at 0x7fa3596f0ee0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/user/opendr/projects/opendr_ws/src/perception/scripts/semantic_segmentation_bisenet.py", line 95, in callback
    self.heatmap_publisher.publish(self.bridge.to_ros_image(heatmap))
  File "/home/user/opendr/projects/opendr_ws/src/ros_bridge/src/opendr_bridge/bridge.py", line 70, in to_ros_image
    message = self._cv_bridge.cv2_to_imgmsg(image.opencv(), encoding=encoding)
AttributeError: 'Heatmap' object has no attribute 'opencv'
```

tsampazk, 2022-09-13:
Oh ok this was a known issue in development that was fixed by Niclas. I think you should update your develop branch as the `opencv` method was added on the `Heatmap` target class 8 days ago as seen [here](https://github.com/opendr-eu/opendr/commit/f59deb6f2a076f7247fddd9a9f9ff673f05029ea#diff-dfcd13e5b831454a1ff9051d09e618e6f9bd5949c121831a3c8cd9a3d0908bfd) and is now present in the [develop branch](https://github.com/opendr-eu/opendr/blob/develop/src/opendr/engine/target.py#L1075-L1081).

Give it a try and let me know so we can close this issue.

thomaspeyrucain, 2022-09-13:
Yes, it is fixing the issue thanks,
Regarding the Heatmap, it is a bit dark is it normal ?
[image]

tsampazk, 2022-09-13:
Yes, this is the image with every pixel value replaced with a class id, so it is supposed to be read and used in code as far as i can tell. 

For visualization purposes i have created an rgb image output on topic `/opendr/heatmap_visualization` that blends the original image with the rgb heatmap (where every color matches a class id) as well as a color-class legend.

thomaspeyrucain, 2022-09-13:
Ok perfect thanks ^^

closed 2022-09-13 by thomaspeyrucain