opendr-eu/opendr #302: rosnode - video_activity_recognition.py - parameters + error

thomaspeyrucain, 2022-09-12:
Parameters need to be consistent with other tools with agparser

I am using opendr installed on my computer on the develop branch
I get this error when launching the node:
```
rosrun perception video_activity_recognition.py 
/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torchvision/transforms/_functional_video.py:5: UserWarning: The _functional_video module is deprecated. Please use the functional module instead.
  warnings.warn(
/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torchvision/transforms/_transforms_video.py:25: UserWarning: The _transforms_video module is deprecated. Please use the transforms module instead.
  warnings.warn(
GPU found.
Padding along the temporal dimension only affects the computation in `forward3d`. In `forward` it is omitted.
/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/video_activity_recognition.py:167: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  return Image(frame, dtype=np.float)
[ERROR] [1662994794.298464]: bad callback: <bound method HumanActivityRecognitionNode.callback of <__main__.HumanActivityRecognitionNode object at 0x7fa3c165ef40>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/video_activity_recognition.py", line 110, in callback
    result = self.learner.infer(x)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/cox3d_learner.py", line 140, in infer
    results = self.model.forward(batch)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/cox3d.py", line 1067, in forward
    x = module(x)
  File "/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/cox3d.py", line 650, in forward
    x = self.avg_pool(x)
  File "/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/pooling.py", line 111, in forward
    output, (self.state_buffer, self.state_index) = self._forward(
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/pooling.py", line 121, in _forward
    pooled_frame = super(RePooled, self).forward(input)
  File "/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torch/nn/modules/pooling.py", line 615, in forward
    return F.avg_pool2d(input, self.kernel_size, self.stride,
RuntimeError: Given input size: (432x6x7). Calculated output size: (432x0x1). Output size is too small

[ERROR] [1662994794.325685]: bad callback: <bound method HumanActivityRecognitionNode.callback of <__main__.HumanActivityRecognitionNode object at 0x7fa3c165ef40>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/video_activity_recognition.py", line 110, in callback
    result = self.learner.infer(x)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/cox3d_learner.py", line 140, in infer
    results = self.model.forward(batch)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/cox3d.py", line 1067, in forward
    x = module(x)
  File "/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/cox3d.py", line 650, in forward
    x = self.avg_pool(x)
  File "/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/pooling.py", line 111, in forward
    output, (self.state_buffer, self.state_index) = self._forward(
  File "/home/thomaspeyrucain/opendr/src/opendr/perception/activity_recognition/cox3d/algorithm/pooling.py", line 121, in _forward
    pooled_frame = super(RePooled, self).forward(input)
  File "/home/thomaspeyrucain/opendr/venv/lib/python3.8/site-packages/torch/nn/modules/pooling.py", line 615, in forward
    return F.avg_pool2d(input, self.kernel_size, self.stride,
RuntimeError: Given input size: (432x6x7). Calculated output size: (432x0x1). Output size is too small

[INFO] [1662994794.331099]: Human activity recognition node started!

```

tsampazk, 2022-10-18:
Hey @thomaspeyrucain could you please check if the issues are indeed resolved (should be, by https://github.com/opendr-eu/opendr/pull/322) in the develop branch, so we can go ahead and close this issue?

thomaspeyrucain, 2022-10-18:
Hello @tsampazk, Yes now it is working, is there a list also of all the activities with the ID ?

LukasHedegaard, 2022-10-18:
I'm not sure what your question is, @thomaspeyrucain. There's a list mapping class ID to human-readable names in `opendr/src/opendr/perception/activity_recognition/datasets/kinetics400_classes.csv`

thomaspeyrucain, 2022-10-18:
Hello @LukasHedegaard ,
Would it be possible to add a link to this file in the documentation of the rosnode?

tsampazk, 2022-10-18:
I think that it would be better to add a link to the .csv file in the learner docs [here](https://github.com/opendr-eu/opendr/blob/develop/docs/reference/activity-recognition.md), as the table is quite big. I will add it there as well as in the ROS docs through the open [PR that overhauls the ROS docs](#316).

closed 2022-10-18 by tsampazk