opendr-eu/opendr #304: rosnode - skeleton_based_action_recognition.py - not executable + message error

thomaspeyrucain, 2022-09-12:
File is not executable

Parameters need to be consistent with other tools with agparser

I am using opendr installed on my computer on the develop branch
```
rosrun perception skeleton_based_action_recognition.py 
Tokenization took: 0.01 ms
Type conversion took: 0.08 ms
Parser memory cleanup took: 0.00 ms
Tokenization took: 0.03 ms
Type conversion took: 0.14 ms
Parser memory cleanup took: 0.00 ms
GPU found.
Downloading pretrained model...
Metadata json file already exists.
Trained model .pth file already exists.
Pretrained model download complete.
Downloading pretrained model...
Downloaded metadata json.
Pretrained model download complete.
Loading checkpoint
Loaded Pytorch model.
[INFO] [1662996169.044649]: Skeleton-based action recognition node started!
[ERROR] [1662996169.134778]: bad callback: <bound method SkeletonActionRecognitionNode.callback of <__main__.SkeletonActionRecognitionNode object at 0x7fd020885f40>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/skeleton_based_action_recognition.py", line 141, in callback
    self.pose_publisher.publish(ros_pose)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 879, in publish
    data = args_kwds_to_message(self.data_class, args, kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msg.py", line 121, in args_kwds_to_message
    raise TypeError("expected [%s] but got [%s]"%(data_class._slot_types[0], arg._type))
TypeError: expected [std_msgs/Header] but got [ros_bridge/OpenDRPose2D]

[ERROR] [1662996169.174963]: bad callback: <bound method SkeletonActionRecognitionNode.callback of <__main__.SkeletonActionRecognitionNode object at 0x7fd020885f40>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/skeleton_based_action_recognition.py", line 133, in callback
    poses = _select_2_poses(poses)
  File "/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/skeleton_based_action_recognition.py", line 174, in _select_2_poses
    selected_poses.append(poses[index])
TypeError: only integer scalar arrays can be converted to a scalar index

[ERROR] [1662996169.199651]: bad callback: <bound method SkeletonActionRecognitionNode.callback of <__main__.SkeletonActionRecognitionNode object at 0x7fd020885f40>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/thomaspeyrucain/opendr/projects/opendr_ws/src/perception/scripts/skeleton_based_action_recognition.py", line 141, in callback
    self.pose_publisher.publish(ros_pose)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 879, in publish
    data = args_kwds_to_message(self.data_class, args, kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msg.py", line 121, in args_kwds_to_message
    raise TypeError("expected [%s] but got [%s]"%(data_class._slot_types[0], arg._type))
TypeError: expected [std_msgs/Header] but got [ros_bridge/OpenDRPose2D]
```

The pose message from pose_estimation has changed

tsampazk, 2022-10-24:
This issue should be fixed by #308

tsampazk, 2022-10-25:
Hey @thomaspeyrucain, could you test the updated relevant node to see if the issues faced persist or not so we can go ahead and close this issue?

thomaspeyrucain, 2022-10-25:
Hello @tsampazk, I just tested it and it is working the same way as before ^^

closed 2022-10-25 by thomaspeyrucain

tsampazk, 2022-10-25:
Great thanks!