opendr-eu/opendr #305: [Tracker] ROS1 nodes updates and dixes

tsampazk, 2022-09-13:
This issue serves to track the progress on the updates on all ROS1 nodes, as per the use-case partner feedback as well as fixes regarding consistency across all nodes. Also, the delay issues mentioned in #275 should be fixed along with the other changes, as well as issue #209 relating to input topics etc.

Main issues to fix:
----

Below there's a list of main issues that need fixing, i tried to check out all the nodes and made some comments in parentheses on what i noticed, so use them to get a general sense on the issues:

1. **Argparse usage** as seen [here](https://github.com/opendr-eu/opendr/blob/develop/projects/opendr_ws/src/perception/scripts/object_detection_2d_centernet.py#L105-L115).
2. **Argument names and default values consistency across nodes**, as seen in [node class ctor](https://github.com/opendr-eu/opendr/blob/develop/projects/opendr_ws/src/perception/scripts/object_detection_2d_centernet.py#L31-L33) and [argparse](https://github.com/opendr-eu/opendr/blob/develop/projects/opendr_ws/src/perception/scripts/object_detection_2d_centernet.py#L105-L115). Arguments especially in argparse which is the interface the users interact with, should have descriptive names like `input_rgb_image_topic` and `input_depth_image_topic`, etc, as well as default values to run out-of-the-box if possible.
3. Updated **device selection** as seen [here](https://github.com/opendr-eu/opendr/blob/develop/projects/opendr_ws/src/perception/scripts/object_detection_2d_centernet.py#L117-L128).
4. Preferably, if the node has multiple outputs, we should provide the option to disable them as seen [here](https://github.com/opendr-eu/opendr/blob/develop/projects/opendr_ws/src/perception/scripts/object_detection_2d_centernet.py#L51-L59), with appropriate changes in the callback too.
5. Make sure you update your docstrings for wrong argument names, typos, some nodes mentioning different tools, etc.
6. Please test whether your node has delay issues as described in #275 and if you need to adjust the queue sizes.
7. Keep in mind the feedback provided by the use-case partners

Tracking list:
----

- [ ] Audiovisual emotion recognition (updates in argparse, device code, device argument, argument naming/default values)
- [ ] Face detection retinaface
- [ ] Face recognition
- [ ] Fall detection
- [ ] Heart anomaly detection (updates in argparse, device code, device argument, docstrings)
- [ ] Landmark-based facial expression recognition (argparse, device code, argument naming/default values, docstrings, `pose_estimation` variables)
- [ ] Object detection 2D - centernet
- [ ] Object detection 2D - detr (some updates in #296, should be good to go)
- [ ] Object detection 2D - gem (argparse, device code, argument naming/default values) (updates in #295)
- [ ] Object detection 2D - ssd
- [ ] Object detection 2D - yolov3
- [ ] Object detection 3D - voxel (argparse, device code, argument naming/default values, docstrings)
- [ ] Object tracking 2D - deep sort (argparse, device code, argument naming/default values)
- [ ] Object tracking 2D - fair mot (argparse, device code, argument naming/default values)
- [ ] Object tracking 3D - ab3dmot (argparse, device code, argument naming/default values, docstrings)
- [ ] Panoptic segmentation efficient ps
- [ ] Pose estimation
- [ ] RGBD hand gesture recognition (argparse, device code, argument naming/default values, docstrings, see #303)
- [ ] Semantic segmentation bisenet
- [ ] Skeleton-based action recognition (argparse, device code, argument naming/default values, see #304)
- [ ] Speech command recognition (updates in argparse, device code, argument naming/default values, docstring)
- [ ] Video activity recognition (argparse, device code, argument naming/default values, docstring, see #302)

negarhdr, 2022-09-13:
Two new branches (on top of develop) for skeleton-based_action_recognition and landmark-based_facial_expression_recognition are created which are called "har_ros1_fix", "fer_ros1_fix", respectively. 
In "har_ros1_fix", the script "skeleton_based_action_recognition" is updated and in  "fer_ros1_fix", the script "landmark_based_facial_expression_recognition" is updated. 
Please check them out and let me know if there are still problems.

tsampazk, 2022-09-14:
Hey @negarhdr! 

I checked the changes out, they seem pretty good apart from some really minor things. 

Feel free to open pull requests to merge them into the develop branch and i can comment on the minor issues on the review.

PR #364 (2022-12-06): ROS1 final updates
This PR should conclude the fixes for the ROS1 workspace and packages of OpenDR.

1. Applied some fixes i missed on the previous PR #357 
2. Renamed all ROS1 package directories to include "opendr_" prefix in their names, e.g. "perception"->"opendr_perception" as suggested
3. Renamed references that i could find to the package names in their corresponding package.xml, CMakeLists.txt, etc.
4. READMEs for opendr_perception package will be updated in a separate PR (#316) to include changes from this PR
5. Added "_node" in all node script filenames, to match those in ROS2 (#256) [... cut]