opendr-eu/opendr #380: ROS1 Object Tracking 2D DeepSort error with input from webcam

tsampazk, 2022-12-14:
I was unable to find it documented so i am opening a new issue with the following error for the deepsort ROS1 node:

```
[ERROR] [1671020211.255272]: bad callback: <bound method ObjectTracking2DDeepSortNode.callback of <__main__.ObjectTracking2DDeepSortNode object at 0x7edf5c2f28>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opendr/projects/opendr_ws/src/opendr_perception/scripts/object_tracking_2d_deep_sort_node.py", line 105, in callback
    tracking_boxes = self.learner.infer(image_with_detections, swap_left_top=True)
  File "/opendr/src/opendr/perception/object_tracking_2d/deep_sort/object_tracking_2d_deep_sort_learner.py", line 289, in infer
    result = self.tracker.infer(image, frame_id, swap_left_top=swap_left_top)
  File "/opendr/src/opendr/perception/object_tracking_2d/deep_sort/algorithm/deep_sort_tracker.py", line 81, in infer
    bbox_xywh[:, 3:] *= 1.2
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
```

What i found the node works properly when provided with images from the `image_dataset_node` but throws this error when taking input from a webcam.

tsampazk, 2022-12-14:
@iliiliiliili could you please take a look at this?

iliiliiliili, 2022-12-19:
Hi @tsampazk, thank you for noticing it. This is a bit hard to check, as I do the tests on headless workstations, but it may be due to no person detections, so I will try that.

closed 2023-09-18 by tsampazk