opendr-eu/opendr #399: ROS1 object 3D detection/tracking nodes still use deprecated pointcloud1

torydebra, 2023-01-20:
Hi,

I was trying to run the ROS nodes for object detection and tracking (`object_detection_3d_voxel_node` and `object_tracking_3d_ab3dmot_node`) but they expect a `sensor_msgs.msg PointCloud` which is a bit deprecated in favour of  `PointCloud2`. 

For example the realsense D400 cameras publish PointCloud2 messages

Is there a way to have support for PointCloud2 as well (or instead) ?

Thanks!

tsampazk, 2023-01-20:
PointCloud2 support will be added in a PR that's a work in progress currently (https://github.com/opendr-eu/opendr/pull/359). After that we can look into using PointCloud2 for the existing nodes.

In the meantime you can take a look in the PR and maybe grab the [bridge methods](https://github.com/opendr-eu/opendr/blob/a6ffb98d131bcf08b75365a3e57ef238ff04afee/projects/opendr_ws/src/opendr_bridge/src/opendr_bridge/bridge.py#L605-L672) from there and modify the existing nodes to work with PointCloud2. 

I can't guarantee that it will be compatible with the learner, etc., @iliiliiliili could maybe provide some more insight on the matter.

iliiliiliili, 2023-01-23:
Hi, I did not find in the official documentation that PointCloud is deprecated and will be replaced by PointCloud2.
These are just different data structures, as PointCloud is intended to be used to store unstructured point clouds only (such as from the Velodyne lidar), while the PointCloud2 data type is intended to be used with point clouds, structured in a 2D grid (such as from an O3M Lidar), but it can also represent unstructured point clouds if we set one of the dimensions as 1.

Voxel-based 3D detection methods that we implemented support only the Velodyne-like Lidars with 360 degrees FOV and 64 beams, and therefore they cannot work with any PointCloud2 data that is incompatible with PointCloud, which means that adding the support to PointCloud2 can make people believe that their sensors can be compatible, when, in fact, they are not.

torydebra, 2023-01-23:
Hi, thanks for the fast responses

About deprecation, I may have misundertood [ROS1](http://wiki.ros.org/pcl/Overview) doc about Pointcloud messages, as it seems PointCloud1 is an "old" stuff. Difference was not so clear for me so in any case thanks for the clarification.
By the way for ROS2 I found deprecation comment for pointcloud1: https://docs.ros2.org/foxy/api/sensor_msgs/msg/PointCloud.html

Is also `object_tracking_3d_ab3dmot_node` based on velodyne-voxel type of pointcloud? In my application I would like to use a Realsense D435 depth camera, can I use these nodes or any of the other nodes for 3D pose estimation?

iliiliiliili, 2023-01-23:
@torydebra 
AB3DMOT works on top of any 3D detections, so if you have a detector that works with your lidar, you can perform tracking on top of it without any problems, as long as the input is a `BoundingBox3DList`.
`Realsense D435 depth camera` is a 2D-like sensor; I had no success on using such sensors (I tried an O3M Lidar) with the pretrained Voxel-based 3D detectors, but it still has a chance to be trained using this sensor.

Yes, ROS2 contains the deprecation warning, so it may be better to switch to PointCloud2 with `height=1`, but we have to ensure that we have enough warnings about the type of data the method expects.

passalis, 2024-05-10:
No update after one year. I currently closing this issue.

closed 2024-05-10 by passalis