opendr-eu/opendr #407: Possible conflict between EfficientLPS & EfficientPS

ad-daniel, 2023-02-13:
**Describe the Bug**

When installing the toolkit from the develop branch I get this error:

```
ERROR: Cannot install geffnet 1.0.1 (from /home/daniel/opendr/src/opendr/perception/panoptic_segmentation/efficient_lps/algorithm/EfficientLPS/efficientNet) and geffnet 1.0.1 (from /home/daniel/opendr/src/opendr/perception/panoptic_segmentation/efficient_ps/algorithm/EfficientPS/efficientNet) because these package versions have conflicting dependencies.

The conflict is caused by:
    The user requested geffnet 1.0.1 (from /home/daniel/opendr/src/opendr/perception/panoptic_segmentation/efficient_lps/algorithm/EfficientLPS/efficientNet)
    The user requested geffnet 1.0.1 (from /home/daniel/opendr/src/opendr/perception/panoptic_segmentation/efficient_ps/algorithm/EfficientPS/efficientNet)

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
```

Which is very odd since they're the same version. @vniclas @aselimc Is there a difference between the two tools that could cause this problem?

vniclas, 2023-02-13:
I believe they are the same, i.e., in one of the requirements files we should just point to the other location. @aselimc will look into this tomorrow.

aselimc, 2023-02-13:
I think they are the same as well. I will try to solve this tomorrow. Sorry for the inconvenience

aselimc, 2023-02-14:
@ad-daniel Can we make assumption that if a user wants to use our toolkit, all of our submodules are needed to be fetched ? Since both EfficientPS and EfficientLPS uses geffnet 1.0.1, I can simply point the other location (which solves the problem) but this will lead to an issue if user only wants to use one of these tools and not install the other one at all.

ad-daniel, 2023-02-14:
The toolkit can be installed as a whole or by modules. However in the case of those two, both tools belong to the same module namely `perception/panoptic_segmentation`. As such yes, I think since you can't currently install one without the other, therefore I think you can go ahead with that assumption and fix.
Thanks for looking into this!

aselimc, 2023-02-14:
I think we can have one more option (because even though EfficientPS and EfficientLPS are both in `perceoption/panoptic_segmentation`, they can be both used seperately).  I can also add the following into `dependencies/parse_dependencies.py`, which will prevent adding the same dependency into wheel. The modified code inside the file `dependencies/parse_dependencies.py` will look like this:

```
global flag_efficientNet
flag_efficientNet = False

python_prerequisites_file = "python_prerequisites.txt"
python_file = "python_dependencies.txt"
linux_file = "linux_dependencies.txt"

def read_ini(path):
    opendr_device = os.environ.get('OPENDR_DEVICE', default='cpu')
    parser = ConfigParser()
    parser.read(path)
    def read_ini_key(key, summary_file):
        tool_device = parser.get('device', 'opendr_device', fallback='cpu')
        if opendr_device == 'cpu' and tool_device == 'gpu':
            return
        if parser.has_option(section, key):
            dependencies = parser.get(section, key)
            if dependencies:
                for package in dependencies.split('\n'):
                    if "efficientNet" in package:
                        global flag_efficientNet
                        if flag_efficientNet:
                            continue
                        else:
                            flag_efficientNet = True
                    with open(summary_file, "a") as f:
                        f.write(os.path.expandvars(package) + '\n')
    read_ini_key('python-dependencies', python_prerequisites_file)
    read_ini_key('python', python_file)
    read_ini_key('linux', linux_file)
```

Would you suggest to go with the assumption or this fix @ad-daniel ?

By the way sorry for the long texts

ad-daniel, 2023-02-14:
Good point, I think changing the `parse_dependencies.py` to include it only once is a good idea. It seems to do the trick on my machine so you can go ahead with that patch. By the way, `install_nvidia.sh` needs to be adapted accordingly too

closed 2023-02-16 by aselimc