opendr-eu/opendr #437: Onnx inference provides confidence score greater than 1.

lakpa-tamang9, 2023-05-26:
I am implementing graph convolution based action recognition and while inferencing with pytorch, the summation of confidence scores of all classes equate to 1. However, while inferencing same model with onnxruntime, the confidence scores of each class is grater than 1.
https://github.com/opendr-eu/opendr/blob/d8b1572e742bab6a2edc5ef8195cbf487c042a4d/src/opendr/perception/skeleton_based_action_recognition/spatio_temporal_gcn_learner.py#L725

passalis, 2023-05-28:
Hi @lakpa-tamang9 

Thank you for your interest in the toolkit and for noticing this. We are looking into this and we will have an update.

tsampazk, 2023-05-29:
Hey @lakpa-tamang9, can you provide more information regarding your issue:

1. Are you running the [provided demo](https://github.com/opendr-eu/opendr/blob/d8b1572e742bab6a2edc5ef8195cbf487c042a4d/projects/perception/skeleton_based_action_recognition/demos/demo.py#L127) with the --onnx argument?
2. What are you printing and getting confidences?
3. Any code that can help us reproduce the issue is welcome.

Maybe @negarhdr has some insight into this?

negarhdr, 2023-05-30:
Hi @lakpa-tamang9, 

Following the information asked by @tsampazk, can you also tell us if you have trained a multi_stream model or not and if so, how did you ensemble the results?

passalis, 2023-07-03:
Is there any update on this issue? Have we managed to reproduce this issue?

passalis, 2023-11-30:
Closing as stale/not able to be reproduced. We can re-open it when more information is available.

closed 2023-11-30 by passalis