opendr-eu/opendr #444: Guidance Request on Obtaining 106_index.npy File

Ted-developer, 2023-07-03:
hello,

I trust this message finds you well.

I am currently working with your code and have come across a requirement for the '106_index.npy' file. However, I am unsure of how to procure this particular file.

I would greatly appreciate any guidance or assistance you might be able to provide on how to obtain the '106_index.npy' file.

Thank you very much for your time and help. I am looking forward to hearing from you soon.

Best regards

passalis, 2023-07-03:
Hi, thank you for you issue and interest in our toolkit. Please provide the steps needed for reproducing the issue that you are facing.

passalis, 2023-11-30:
Closing as stale. We can re-open when more information regarding reproducing the issue are provided.

closed 2023-11-30 by passalis