opendr-eu/opendr #449: Face recognition wheel test failure

stefaniapedrazzi, 2023-07-05:
It was working a couple of weeks ago, so it could be due to some dependencies update.
Error log:
```
======================================================================
ERROR: test_fit (test_face_recognition.TestFaceRecognitionLearner)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/opendr/opendr/tests/sources/tools/perception/face_recognition/test_face_recognition.py", line 61, in test_fit
    results = recognizer.fit(dataset=train, silent=False, verbose=True)
  File "/home/runner/work/opendr/opendr/venv/lib/python3.8/site-packages/opendr/perception/face_recognition/face_recognition_learner.py", line 222, in fit
    dataset_train = datasets.ImageFolder(dataset.path, train_transform)
  File "/home/runner/work/opendr/opendr/venv/lib/python3.8/site-packages/torchvision/datasets/folder.py", line 309, in __init__
    super().__init__(
  File "/home/runner/work/opendr/opendr/venv/lib/python3.8/site-packages/torchvision/datasets/folder.py", line 143, in __init__
    super().__init__(root, transform=transform, target_transform=target_transform)
  File "/home/runner/work/opendr/opendr/venv/lib/python3.8/site-packages/torchvision/datasets/vision.py", line 39, in __init__
    if isinstance(root, torch._six.string_classes):
AttributeError: module 'torch' has no attribute '_six'
``

Pavlos-Tosidis, 2023-07-05:
This is most certain due to missmatching cuda installation of torch and torch vision.

https://github.com/opendr-eu/opendr/blob/b89ea14dfd64341680f67b0650d8d84016649456/bin/install.sh#L66

This should be changed to provide cuda version for torchvision as well, since it gets installed with cu117, into:

python3 -m pip install torch==1.13.1+cu116 torchvision==0.14.1+cu116 torchaudio==0.13.1 -f https://download.pytorch.org/whl/torch_stable.html

or according to pytorch instructions:

python3 -m pip install torch==1.13.1+cu116 torchvision==0.14.1+cu116 torchaudio==0.13.1 --extra-index-url https://download.pytorch.org/whl/cu116

stefaniapedrazzi, 2023-07-11:
It seems that dependencies have been been fixed.
In the latest scheduled test the error is solved.

closed 2023-07-11 by stefaniapedrazzi