opendr-eu/opendr #453: missing unzip dependency for install script

thomaspeyrucain, 2023-08-16:
If unzip is not installed the installation script fails with this error:
```
Torch version not defined, version 1.9.0 will be installed.
For a specific Torch version please define TORCH_VERSION with 'export TORCH_VERSION=x.x.x'
Downloading and installing LibTorch and torchvision (cpu-only) ...
Downloading LibTorch and torchvision ...
Downloading Libtorch and torchvision done.
./install_torch_c_api.sh: line 37: unzip: command not found
./install_torch_c_api.sh: line 38: cd: libtorch: No such file or directory
tar (child): vision.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
mv: cannot stat 'vision-*': No such file or directory
./install_torch_c_api.sh: line 47: cd: vision: No such file or directory
CMake Error: The source directory "/home/user/exchange/opendr" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
make[1]: *** No rule to make target 'desktop'.  Stop.
make: *** [Makefile:42: install_compilation_dependencies] Error 2
```

tsampazk, 2023-08-17:
Hey @thomaspeyrucain, thanks for reporting this. Could you provide some additional information?

- Are you attempting a full toolkit installation on a desktop by cloning and running the installation script as described [here](https://github.com/opendr-eu/opendr/blob/master/docs/reference/installation.md#installing-by-cloning-opendr-repository-ubuntu-2004-x86-architecture)?
- After failing, did you install unzip, then retried the installation and it ran successfully? If so, how exactly did you install unzip?

tsampazk, 2023-08-17:
Thanks! It seems that most systems that we use have unzip pre-installed. I think we can install unzip [along other base ubuntu dependencies](https://github.com/opendr-eu/opendr/blob/b89ea14dfd64341680f67b0650d8d84016649456/bin/install.sh#L21), which should do the trick.

PR #454 (2023-08-24): Added unzip installation as base ubuntu dependency and tool tests fixes
Fixes: 
- Possible fix #453.
- Replaced `bcolz` with `bcolz-zipline` to fix broken installation
- [In tests_suite.yml](https://github.com/opendr-eu/opendr/blob/e3ed0a2117a31e747b1c087896045d01f0e89f5d/.github/workflows/tests_suite.yml#L105) fixed a call for `pip install /tests/sources/requirements.txt` which is an empty file to `pip install /tests/requirements.txt` which seems to be the correct one
- More strict pillow version, added <10.0.0, which breaks calls to `LINEAR` from detectron2 (which itself sits at an older ver [... cut]
closed 2023-08-24 by tsampazk