opendr-eu/opendr #457: docker image from v2.2.0 - can't load  from opendr.perception.object_detection_2d import YOLOv5DetectorLearner

Niknu, 2023-08-28:
When I try load `from opendr.perception.object_detection_2d import YOLOv5DetectorLearner` from in the docker image from v2.2.0 I get this error:

`File "openDr_show_custom_works.py", line 7, in <module>
    from opendr.perception.object_detection_2d import YOLOv5DetectorLearner
  File "/opendr/src/opendr/perception/object_detection_2d/__init__.py", line 1, in <module>
    from opendr.perception.object_detection_2d.centernet.centernet_learner import CenterNetDetectorLearner
  File "/opendr/src/opendr/perception/object_detection_2d/centernet/centernet_learner.py", line 25, in <module>
    from gluoncv.data.transforms import presets
ModuleNotFoundError: No module named 'gluoncv'`

passalis, 2023-08-29:
Thanks for catching this. A dependency issue broke the automated CUDA container building process #458 

Can you try using this custom image (`opendr/opendr-toolkit:cuda_nightly_290823`) can verify if this resolves this issue?

passalis, 2023-08-30:
Can you share the exacts commands for reproducing the issue? Have you installed `nvidia-docker` (https://github.com/opendr-eu/opendr/blob/master/docs/reference/installation.md#gpu-docker) ?

I am running the following and I am not getting any error:
```
sudo docker run --gpus all -it opendr/opendr-toolkit:cuda_nightly_290823  /bin/bash
source ./bin/activate.sh 
```

tsampazk, 2023-08-30:
I can confirm that what Nikos suggests runs fine on my end too. I also tried to run some demos and i'm getting qt display errors which are to be expected.

Niknu, 2023-08-30:
This is not the gpu but the cpu docker image

passalis, 2023-08-30:
Can you please try this one `opendr/opendr-toolkit:cpu_nightly_300823`?

Niknu, 2023-08-31:
> Can you please try this one `opendr/opendr-toolkit:cpu_nightly_300823`?

This works now for me, thanks

Niknu, 2023-08-31:
Sorry for open this one again, but after digging and trying out the docker image `opendr/opendr-toolkit:cpu_nightly_300823` I can't be able to display a picture. I have tried the run my own python script in the docker where the X11 is mount as volume (`-v /tmp/.X11-unix:/tmp/.X11-unix`) and also tried the with the jupyter notebook. I have tried to run this code:

```
import torch
from opendr.engine.data import Image # NOTE: Image load cv2, and can't find it, have call "source bin/activate.sh" mentioned here: https://github.com/opendr-eu/opendr/blob/v2.2.0/docs/reference/installation.md
from opendr.perception.object_detection_2d import YOLOv5DetectorLearner
from opendr.perception.object_detection_2d import draw_bounding_boxes

yolo = YOLOv5DetectorLearner(model_name='custom', path='/mount_deep_learning/weight/june2023v2.pt' , device='cpu')

torch.hub.download_url_to_file('https://ultralytics.com/images/zidane.jpg', 'zidane.jpg')  # download image
im1 = Image.open('zidane.jpg')  # OpenDR image

results = yolo.infer(im1)
draw_bounding_boxes(im1.opencv(), results, yolo.classes, show=True, line_thickness=3)
```

passalis, 2023-08-31:
This is caused by opencv and getting opencv to work correctly inside docker can be tricky (i.e., the issue does not arise due to OpenDR). The following works for me (but I am not sure if this will work for all systems):
```
sudo docker run -it -v /tmp/.X11-unix:/tmp/.X11-unix -v "$HOME/.Xauthority:/root/.Xauthority:ro" --network host -e DISPLAY=unix$DISPLAY opendr/opendr-toolkit:cpu_nightly /bin/bash
```
However, the easiest solution would be to just disable visualization, i.e., set `show=False` in `draw_bounding_boxes` and this should work in any case. Then, you can just save the image and examine it.

closed 2023-08-31 by Niknu