opendr-eu/opendr #458: CUDA docker images are not built correctly

passalis, 2023-08-29:
After the last CUDA version bump, the CUDA docker image is no longer correctly built due to conflicting dependencies.
The issue arises due to an mmdet conflict that is triggered due to the version used to compile the installed pytorch version and the installed CUDA toolkit version. Updating the CUDA version is not easy at the moment, since this will break other tools, while there is not pytorch binary precompiled for the current CUDA version (11.2).

So, at the moment, a (possible) workaround is the following:

- Start from an existing image
- Remove the dependencies that cause the issues
```
rm src/opendr/perception/panoptic_segmentation/efficient_ps/dependencies.ini 
rm src/opendr/perception/panoptic_segmentation/efficient_lps/dependencies.ini 
```

- Re-install the toolkit:
``` ./bin./install.sh ```

I will uploaded a nightly docker image to verify that this works and then see how this issues can be tackled.

tsampazk, 2023-08-30:
We should probably check for similar issues in the CPU docker image.

passalis, 2023-08-30:
The existing image is indeed broken, but the updated master seems to lead to cpu images that work as expected. I am uploading an updated version.

closed 2023-11-30 by passalis