opendr-eu/opendr #460: argparse issue with launchfiles

thomaspeyrucain, 2023-09-04:
Hello,

If I want to integrate the nodes in a launchfile like this:
```
<?xml version="1.0" encoding="UTF-8"?>
<launch>
    <arg name="rgb_topic" default="/xtion/rgb/image_raw"/>
    <node name="face_reco_node" pkg="opendr_perception" type="face_recognition_node.py" args="-i $(arg rgb_topic) -dataset_path $(arg face_reco_path)" output="screen"/> 

</launch>
```

I get this error:

```
opendr_ws$ roslaunch opendr_perception opendr_demo.launch 
... logging to /home/thomaspeyrucain/.ros/log/0e98f1ec-4afe-11ee-b07b-68545ace404b/roslaunch-manama-16957.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.6.172:39345/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    face_reco_node (opendr_perception/face_recognition_node.py)

ROS_MASTER_URI=http://tiago-145c:11311

process[face_reco_node-1]: started with pid [16980]
usage: face_recognition_node.py [-h] [-i INPUT_RGB_IMAGE_TOPIC] [-o OUTPUT_RGB_IMAGE_TOPIC] [-d DETECTIONS_TOPIC]
                                [-id DETECTIONS_ID_TOPIC] [--performance_topic PERFORMANCE_TOPIC] [--device {cuda,cpu}]
                                [--backbone {mobilefacenet}] [--dataset_path DATASET_PATH]
face_recognition_node.py: error: unrecognized arguments: __name:=face_reco_node __log:=/home/thomaspeyrucain/.ros/log/0e98f1ec-4afe-11ee-b07b-68545ace404b/face_reco_node-1.log
INFO:rospy.core:signal_shutdown [atexit]
[face_reco_node-1] process has died [pid 16980, exit code 2, cmd /home/thomaspeyrucain/Desktop/OpenDR/opendr/projects/opendr_ws/src/opendr_perception/scripts/face_recognition_node.py -i /xtion/rgb/image_raw __name:=face_reco_node __log:=/home/thomaspeyrucain/.ros/log/0e98f1ec-4afe-11ee-b07b-68545ace404b/face_reco_node-1.log].
log file: /home/thomaspeyrucain/.ros/log/0e98f1ec-4afe-11ee-b07b-68545ace404b/face_reco_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

To fix it I replaced this line: https://github.com/opendr-eu/opendr/blob/b3d6ce670cdf63469fc5766630eb295d67b3d788/projects/opendr_ws/src/opendr_perception/scripts/face_recognition_node.py#L163
By this line: `args = parser.parse_args(rospy.myargv()[1:])`
In all the ROS scripts

tsampazk, 2023-09-04:
Hey @thomaspeyrucain, thank you for reporting this and the fix. I will test this out and get back to you.

tsampazk, 2023-09-22:
Hello again @thomaspeyrucain, i did some testing regarding this issue.

I used the launch file you provided with some minor modifications to fit my system:

```
<?xml version="1.0" encoding="UTF-8"?>
<launch>
    <arg name="rgb_topic" default="/usb_cam/image_raw"/>
    <node name="face_reco_node" pkg="opendr_perception" type="face_recognition_node.py" args="-i $(arg rgb_topic)" output="screen"/> 
</launch>
```

and i got the same error as the one you reported.

Your fix looks perfectly fine to me as it doesn't interfere with non-launch running of the nodes.

The issue originates from the fact that the launch process seems to use the rospy arguments `rospy.myargv()` which includes the node's path, e.g. when i printed it i got `['/home/kostas/dev/opendr/projects/opendr_ws/src/opendr_perception/scripts/face_recognition_node.py', '-i', '/usb_cam/image_raw']`, so your fix of skipping the first element of this list rectifies the issue.

One minor issue i noticed, which you should probably be already aware of, is that the working directory when using the launchfile is not the workspace directory but something similar to `/home/<user>/.ros`, so in this particular node you need to position your `./database` folder appropriately or provide an appropriate path for face recognition to work properly.

I will go ahead and fix this across all ROS1 nodes so they can be used with launch files. Thank you for your efforts!

thomaspeyrucain, 2023-09-22:
Perfect thanks @tsampazk , for the directory I most often put the whole path of the file or folder

PR #465 (2023-09-22): Fix ROS1 nodes argparse issue with .launch files
Closes #460.

See discussion in #460 for more details.

This PR applies the fix suggested by @thomaspeyrucain to all ROS1 nodes. The fix was tested and doesn't interfere with running the nodes normally.
closed 2023-09-22 by tsampazk