opendr-eu/opendr #507: ModuleNotFoundError: No module named '_constants' when importing ColoredRenderer from opendr.renderer

1997MarsRover, 2024-06-12:
I'm experiencing an issue when trying to import the ColoredRenderer from opendr.renderer. I get a ModuleNotFoundError indicating that the _constants module is missing.

## Steps to reproduce:

Install OpenDR using pip: pip install opendr
Try to import the ColoredRenderer from opendr.renderer: from opendr.renderer import ColoredRenderer
Error message:
```bash
ModuleNotFoundError: No module named '_constants'
Environment:
```
Python version: 3.10
OpenDR version: 0.78
Operating System: Windows 10 (using WSL) 

## Additional information:

I've checked the OpenDR GitHub page and couldn't find any similar issues reported.

## Expected behavior:

I expect to be able to import the ColoredRenderer from opendr.renderer without encountering the ModuleNotFoundError.

## Actual behavior:

I get a ModuleNotFoundError indicating that the _constants module is missing.

## Reproduction code:

from opendr.renderer import ColoredRenderer 

## System information:

Python version: 3.10
pip version: 21.2.4
Operating System: Windows 10 (using WSL)
Any other relevant information:

I've tried reinstalling OpenDR, verifying Cython installation, and checking for conflicts with other packages, but the issue persists.

Please let me know if you need any additional information to help resolve this issue.

passalis, 2024-06-12:
Hi, thank you for opening the issue. However, I think you opened the issue in the wrong repository, since this is the Open Deep Robotics Toolkit ([opendr.eu](https://opendr.eu/)) not the open renderer library.

1997MarsRover, 2024-06-12:
@passalis thanks for the clarification.

closed 2024-06-12 by 1997MarsRover